elixir-circuits/circuits_i2c #9: Remove I2C address from open?

fhunleth, 2018-10-30:
This has been bugging me for a while, but to avoid backwards incompatibility with Elixir/ALE I never changed it. You can pass an I2C address to open and the API will remember it. However, you can also call read_device, write_device, and write_read_device and pass a different I2C address in. I'm finding in my own code that having the API remember a default I2C address is not as useful as I thought it would be. 

What does everyone think of removing that feature? I.e., add the I2C address to read, write, and write_read; delete the _device functions and remove the I2C address from open.

mdsebald, 2018-10-30:
I'm good with that.

mattludwigs, 2018-10-30:
Sounds good to me!

mdsebald, 2018-11-01:
I'd like to change my opinion of this change.  

I was updating my personal project, to eliminate the I2C address from open(), but then there were many, many places where I had to add the I2C address as a function parameter, where I was just passing the I2C reference before.   

I think it would ease the transition from ElixirALE to Circuits, if we kept the open() that allows setting a default I2C address, but still I think the idea of dropping the I2C address from open() is good too.   

I propose we have pairs of functions that allow the user to specify the I2C address or not.  We could have open/1 (no address)  and open/2  (sets default address).  read/2, write/2, and write_read/3 which use the default address set by open/2 (like the current API), and read/3, write/3, and write_read/4 which allow the user to specify the address (same as the current *_device() functions)

So really, we would just add  an open/1 function, and remove "_device" from the function names that are currently specifying the I2C address.  

What do you guys think?   It would save me a lot of typing.

fhunleth, 2018-11-01:
How about creating a helper function in your code that adds the address to the calls to I2C.read and I2C.write?

mdsebald, 2018-11-01:
I hadn't thought of that.  Basically, writing code to do the work ElixirALE used to do.  That would work.  I guess I just want to have my cake and eat it too.  :)  If it's more important to keep the API clean, I won't argue.  Thanks for hearing me out.

fhunleth, 2018-11-01:
If you like how your code looks with the helper function, then let's encourage people to do that. 

I'll give you another example of why I like the new API. I have a device that responds to 3 addresses. If I didn't know better, I'd call open 3 times (one for each address) and get 3 references back. That's, of course, silly since the C code is only passing the address as a convenience and there's no reason to open 3 i2c device files, etc. The option that I'm doing is passing 0 as the I2C address in the open call and only calling read_device and write_device. That doesn't feel great either.

fhunleth, 2018-11-01:
I should add that if you really don't like what your code looks like, please push your case. I do think that you bring up a good point in that if we go ahead with this API that we add a note to the PORTING.md file to encourage people to make helper functions for this.

mdsebald, 2018-11-01:
Definitely I want to be able to open a channel without specifying an address, like you, but I want to have it both ways too.   
For the helper function(s), all I can think of creating an ETS table and storing the I2C reference and address in a tuple, on open, and then retrieving the address matching the reference on read/write.  Do you have any simpler way to do this?

fhunleth, 2018-11-01:
Pass a tuple around that has the address? `{i2c_ref, address}` and then using that. Using an ETS table sounds complicated - how do you garbage collect it? Anyway, I'm still not convinced that saving users from writing a one line helper function is too much to ask. Do you have an example piece of code that illustrates the problem?

fhunleth, 2018-11-01:
What I meant was instead of passing around I2C refs, pass around {i2c ref, address} tuples. The code would be:

```elixir
def my_i2c_open(dev, address) do
  with {:ok, ref} = I2C.open(dev), do: {:ok, {ref, address}}
end

def my_i2c_write({ref, address}, data}), do: I2C.write(ref, address, data)

def my_code() do
  mydevice = my_i2c_open("i2c-1", 0x54)
  my_i2c_write(mydevice, <<1,2,3>>)
  my_i2c_write(mydevice, <<4,5,6>>)
end

def another_fun(i2cdev) do
  my_i2c_write(i2cdev,  <<1,2,3>>)
end
```

fhunleth, 2018-11-01:
For the bme280, I think that the following would be better:

```elixir
def bme280_write(i2c, data), do: I2C.write(i2c, @bme280_address, data)
def bme280_read(i2c, data), do: I2C.write(i2c, @bme280_address, data)
```

fhunleth, 2018-11-01:
@mdsebald - I saw your i2c_util code. Interested in #10?

mdsebald, 2018-11-01:
Yes, I see that, but what I was getting at, was that I would still need to touch the code, wherever the I2cRef is replaced with the tuple, because the -spec() and parameter name would change.

I see the bme280 sensor has a constant address, but that is not necessarily true for all I2C devices.

Sure, I would be interested in #10.

fhunleth, 2018-11-01:
You already need to change your `-spec()`'s since circuits_i2c refers to the i2c bus using a reference and not a pid like Elixir/ALE, right? But you probably want your specs to use a custom type to refer to the i2c bus/device so that you only need to change it in one place. The other pretty important change is that read returns {:ok, data} tuples now, so every call will need to be inspected anyway. I don't think that anyone is getting away with not checking every call to ElixirALE when converting their code.

closed 2018-11-02 by fhunleth