elixir-circuits/circuits_i2c #10: Include Erlang bindings?

fhunleth, 2018-11-01:
They seem like they'd be fairly trivial. Maybe something like:

```elixir
defmodule :circuits_i2c do
  defdelegate open(devname), to: Circuits.I2C
  etc.
end
```

mdsebald, 2018-11-02:
Currently, to call `Circuits.I2C.open(...)` from Erlang code, would be:` 'Elixir.Circuits.I2C':open(...)`

What would the Erlang code look like when using this binding?

Could this `defmodule` reside in the i2c.ex file, or is a separate module needed?

We would need to do the same for GPIO and SPI, and update README, right?

fhunleth, 2018-11-02:
For the code I posted above, the Erlang would look like `circuits_i2c:open("i2c-1").` I think the module is short enough that it could live at the bottom of `i2c.ex` for now at least.

fhunleth, 2018-11-07:
Fixed in 2cf7322b00400c9f7ef76973891c0fce67bf9d9b

closed 2018-11-07 by fhunleth