elixir-circuits/circuits_i2c #47: RPi4B :: After bootloader update, previously working app now gets {:error, :i2c_nak} on I2C.write ??

jsimmonds2, 2019-10-24:
Doesn't make a heap of sense on the surface of it !
Just alerting a possible issue on Raspberry Pi 4 Model B in hopes someone else can correlate.

Meantime I'll double check my Elixir coding and try to debug further.

fhunleth, 2019-10-24:
Just to make sure that I understand, the only change was to update the RPi4 boot EEPROM and I2C broke? And it's this version: https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md#2019-09-10---git-f626c772

jsimmonds2, 2019-10-24:
Yes that was the only change and that is the Git hash 👍 
```
iex(fb2@fb2-e53f.local)2> ~c'vcgencmd bootloader_version'|>:os.cmd()|>IO.puts
Sep 10 2019 10:41:50
version f626c772b15ba1b7e0532a8d50a761b3ccbdf3bb (release)
timestamp 1568112110

:ok
iex(fb2@fb2-e53f.local)3>
```
I have a simple breadboarding setup cabled to that RPi4B . . I'll double-check I didn't disturb anything important while I was swapping SDHC cards !

The app has 17 calls to **`I2C.write`** but just one call soon after init seems to be rejected.
I'll recode that call with **`:retries`** non-zero and see whether that helps . . though I dislike arbitrary 
 permanent "band-aids" like that where the behaviour is not well understood :|

@fhunleth, Thanks !

jsimmonds2, 2019-10-25:
@fhunleth, I'm So sorry for wasting your valuable time . . the GND to the other chip on that breadboard ( a DS3231 RTC ) was found floating in the air !

Adding **`[{:retries, 2}]`** to that failing **`I2C.write`** on the LPS25HB device did squelch the error, but the app didn't get started normally, so some other I2C operation likely silently failed to work correctly.

closed 2019-10-25 by jsimmonds2

fhunleth, 2019-10-25:
I'm glad you found it!!!