elixir-circuits/circuits_i2c #75: Detecting nothing connected on any I2C bus

mnishiguchi, 2021-05-15:
I wonder if it is viable to detect nothing connected on any I2C bus.

I noticed that `Circuits.I2C.open` returns the OK tuple for "i2c-1" even if nothing is connected on that bus. Ideally we want to return an error tuple in such a case so that we can alert the library user.

```elixir
iex(livebook_gq7x2c5s@nerves-9bd5)1> Circuits.I2C.open("i2c-1")
{:ok, #Reference<0.856605456.268828681.72219>}

iex(livebook_gq7x2c5s@nerves-9bd5)2> Circuits.I2C.open("i2c-2")
{:error, :bus_not_found}

iex(livebook_gq7x2c5s@nerves-9bd5)3> Circuits.I2C.open("i2c-1")
{:ok, #Reference<0.856605456.268828681.72220>}

iex(livebook_gq7x2c5s@nerves-9bd5)4> Circuits.I2C.bus_names
["i2c-1"]
```

fhunleth, 2021-05-15:
This is by design.

It's ok for nothing to be on the I2C bus when `open/2` is called: devices could be disabled (turned off via a GPIO to save power or reset it), devices could be pluggable (HDMI's I2C is an example of this where a device is only present when the cable is plugged in), or there could just be nothing connected to the bus on purpose (device not populated on a board).

Generic device detection is not an I2C feature. Heuristics exist, but there's no guarantee that they work on all devices and they could put devices into a bad state. In other words, scanning a bus for devices is nice for debugging and getting started. That's why functions exist in this library to do that. However, it's not something that you'd want for prod since you should know your device ID by then and be able to detect it by directly probing it.

mnishiguchi, 2021-05-15:
I see. It makes sense.

closed 2021-05-15 by mnishiguchi